Xinyuan-LilyGO/T-Panel #2: Version?

Bwooce, 2024-02-14:
Got a shortcut so I can tell which board version I have? I assume since mine has no CAN then it's v1 or v1.1?  (I can't see any silkscreen version)

Llgok, 2024-02-29:
Two versions are currently available: T-Panel_V1.0 and T-Panel_V1.2 The hardware of the two versions is different, and the corresponding software is also different. You can find the relevant macro definitions in "./Mylibrary/pin_config.h"

1.T-Panel_V1.0 version only has an RS485 chip. Uncomment "#define T_Panel_V1_0_RS485".
2.T-Panel_V1.2 version can choose either an RS485 chip or a CAN bus chip based on the module inserted on the board. Uncomment "#define T_Panel_V1_2_RS485" or "#define T_Panel_V1_2_CAN" accordingly.

Llgok, 2024-02-29:
> Got a shortcut so I can tell which board version I have? I assume since mine has no CAN then it's v1 or v1.1? (I can't see any silkscreen version)

Two versions are currently available: T-Panel_V1.0 and T-Panel_V1.2 The hardware of the two versions is different, and the corresponding software is also different. You can find the relevant macro definitions in "./Mylibrary/pin_config.h"

1.T-Panel_V1.0 version only has an RS485 chip. Uncomment "#define T_Panel_V1_0_RS485".
2.T-Panel_V1.2 version can choose either an RS485 chip or a CAN bus chip based on the module inserted on the board. Uncomment "#define T_Panel_V1_2_RS485" or "#define T_Panel_V1_2_CAN" accordingly.

Bwooce, 2024-02-29:
Thanks! I have an RS485 version but not sure if it's 1.0 or 1.2, how can I tell? There's no version number written on the physical board?

Llgok, 2024-02-29:
> Thanks! I have an RS485 version but not sure if it's 1.0 or 1.2, how can I tell? There's no version number written on the physical board?

The replaceable long-distance transmission module is version 1.2. 
The non-replaceable long-distance transmission module is version 1.0.

The following diagram is of version 1.2.
[image]

closed 2024-05-07 by Llgok