Xinyuan-LilyGO/T-Panel #7: CAD files available?

shyndman, 2024-08-02:
### IDE and version or firmware

n/a

### Operating System

n/a

### Description

Hello there,

I was wondering if you might release step files for the T-Panel's hardware variants. I'd really like to design some accurate cases for this thing. :)

PS. Thank you for the wonderful device. I'm having a lot of fun with it, and learning a lot. 
PPS. Modelled a little leanback stand for use during development ([Printables link](https://www.printables.com/model/962461-lilygo-t-panel-s3-development-stand)), but without step files it's slightly wonky.

### Sketch

```cpp
n/a
```

### Debug Message

```plain
n/a
```

Llgok, 2024-11-21:
@shyndman 
Updated [T-Panel_V1.2 CAD](https://github.com/Xinyuan-LilyGO/T-Panel/blob/arduino-esp32-libs_V2.0.14/structure/T-Panel_V1.2.dxf) Drawings

Llgok, 2025-01-02:
Closed due to prolonged inactivity.

closed 2025-01-02 by Llgok