Xinyuan-LilyGO/T-Panel #8: Please clarify details on CANFD

TobyLies, 2024-08-07:
### IDE and version or firmware

T-Panel_V1.3 CANFD

### Operating System

T-Panel_V1.3 CANFD

### Description

Can you give some clarification on the CANFD version of the board? Is it CAN (Controller Area Network) or is it CAN FD (Flexible Data)? If it's CANFD, I gather that TWAI isn't compatible with CANFD?

I don't see any other reference to CANFD in the github.

I want to use the T-Panel to read CAN messages in a car, and display info derived from it. Does the T-Panel both send and receive CAN messages? Or does it require an external board?

### Sketch

```cpp
n/a
```

### Debug Message

```plain
n/a
```

aokhrimenko, 2024-09-02:
What CAN transceiver do you use on this board? I see nothing on schematics.

viniciusro, 2024-09-02:
TD501MCANFD

Llgok, 2024-09-03:
> What CAN transceiver do you use on this board? I see nothing on schematics.

@aokhrimenko 
@viniciusro 
The following is the data manual for the CAN module.
[TD501MCANFD_MORNSUN](https://github.com/Xinyuan-LilyGO/T-Panel/blob/arduino-esp32-libs_V2.0.14/information/TD501MCANFD_MORNSUN.pdf)

Llgok, 2024-09-10:
Closed due to lack of response for a long time.