Xinyuan-LilyGO/T-Panel #10: Adding WIFI to Lvgl example will make it crash.

viniciusro, 2024-08-17:
### IDE and version or firmware

framework = arduino, platform = espressif32, board = dfrobot_firebeetle2_esp32s3

### Operating System

Windows 10

### Description

Adding simple wifi connect to setup will make the example crash:

### Sketch

```cpp
`    // Connect to Wi-Fi
    WiFi.mode(WIFI_MODE_STA);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED)
    {
        delay(5);
        Serial.print(".");
    }
    Serial.println("Connected to WiFi");
    Serial.println(WiFi.localIP());`
```

### Debug Message

```plain
Ciallo
.................Connected to WiFi
192.168.188.75
Found xl9535
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42005458  PS      : 0x00060f30  A0      : 0x82005575  A1      : 0x3fcebb50
A2      : 0x3fcebb74  A3      : 0xffffffff  A4      : 0x00000000  A5      : 0x000000ff
A6      : 0xffffffff  A7      : 0x000000ff  A8      : 0x820053b6  A9      : 0x3fcebb30
A10     : 0x00000000  A11     : 0x3fcebbc0  A12     : 0x3fcebbb8  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x3fceba4c  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000017  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x42005455:0x3fcebb50 0x42005572:0x3fcebb70 0x42007473:0x3fcebbb0 0x42007c0b:0x3fcebc60 0x4209faa2:0x3fcebc90 0x420053c7:0x3fcebcb0 0x42005572:0x3fcebcd0 0x4200c5cd:0x3fcebd10 0x4200c699:0x3fcebd60 0x4200c9d2:0x3fcebde0 0x4200cb03:0x3fcebe10 0x4200d5bf:0x3fcebee0 0x420198e8:0x3fcebf30 0x420024c3:0x3fcebf60 0x4202db09:0x3fcebf80

ELF file SHA256: 20011c7cb8b2c78f

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420a232a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
```

### Other steps to Reproduce

After initialization and connection the WIFI is not used in any other place.

VolosR, 2024-08-23:
I have identical problem. Lvgl example will not work if board is connected to wifi.

nikthefix, 2024-08-24:
@viniciusro 

As an experiment try replacing

lv_disp_draw_buf_init(&draw_buf, buf_1, buf_2, 48 * 1024 * 2);
with
lv_disp_draw_buf_init(&draw_buf, buf_1, NULL, 2 * 480 * 10);

nik

viniciusro, 2024-08-25:
> @viniciusro
> 
> As an experiment try replacing
> 
> lv_disp_draw_buf_init(&draw_buf, buf_1, buf_2, 48 * 1024 * 2); with lv_disp_draw_buf_init(&draw_buf, buf_1, NULL, 2 * 480 * 10);
> 
> I don't have this board so I'm speculating here.
> 
> nik

Hey nik,

Thanks for the idea, it works!
Now I can sed UDP packages and display data on the screen.

I will investigate further the second buffer.

```cpp
    // lv_disp_draw_buf_init(&draw_buf, buf_1, buf_2, 48 * 1024 * 2);
    lv_disp_draw_buf_init(&draw_buf, buf_1, NULL, 2 * 480 * 10);
```

- draw_buf -- pointer lv_disp_draw_buf_t variable to initialize

- buf1 -- A buffer to be used by LVGL to draw the image. Always has to specified and can't be NULL. Can be an array allocated by the user. E.g. static lv_color_t disp_buf1[1024 * 10] Or a memory address e.g. in external SRAM

- buf2 -- Optionally specify a second buffer to make image rendering and image flushing (sending to the display) parallel. In the disp_drv->flush you should use DMA or similar hardware to send the image to the display in the background. It lets LVGL to render next frame into the other buffer while previous is being sent. Set to NULL if unused.

- size_in_px_cnt -- size of the buf1 and buf2 in pixel count.

nikthefix, 2024-08-25:
@viniciusro 
Thanks for letting me know that it worked.
I made conservative assumptions as I didn't know how you defined your buf_1 but if you sized it to contain 16 bit/px then you could further reduce the lv draw buffer to just:

 lv_disp_draw_buf_init(&draw_buf, buf_1, NULL, 480 * 10); //single buffer, line px width, number of lines per write.

In my experience 10 to 40 lines per write is good use of resources for the 480x480 display and keeps things 'ticking' over nicely.
I have personally not seen any advantage to using double buffering with st7701 / esp32-s3 but your mileage may vary.

Volos Projects released a video this morning using the T-Panel:

https://www.youtube.com/watch?v=6-7ZjbE4Qqc

nik

Llgok, 2024-09-10:
Closed due to lack of response for a long time.

closed 2024-09-10 by Llgok