Xinyuan-LilyGO/T-Panel #13: LVGL Switch click not working properly

8lue5ky, 2024-09-08:
### IDE and version or firmware

PlatformIO 6.1.15/3.4.4

### Operating System

Windows10

### Description

When using your LVGL file from the examples template as base with my own UI export files from square line studio, the toggle switch is somehow not working every time when clicked.

I made a small recording of the behavior:

[image]

It looks like the button is sometime clicked multiple times when touched once.

I saw that VolosR added some kind of debounce mechanism in his project [TPanelNetwork](https://github.com/VolosR/TPanelNetwork/blob/main/Panel/Panel.ino). But with that workaround, some other controls (e.g. Arc) are not really usable anymore.

Do you have any ideas how to improve this?

### Sketch

```cpp
Lvgl from examples folder with a toggle button.
```

### Debug Message

```plain
None.
```

Llgok, 2024-09-10:
@8lue5ky 
Hello, could you provide the code for the touch callback function in your LVGL project? Or, you could share the entire original file of your project, which would be more conducive to troubleshooting and resolving our issues.

8lue5ky, 2024-09-10:
I creates a copy of the Lvgl example with a screen just containing a toggle switch and a arc.

[Commit with example](https://github.com/8lue5ky/T-Panel/commit/4f4b5f497aa39810fa4b51a0713a6ffe2b73741b)

It should look like this:
[image]

Llgok, 2024-09-11:
@8lue5ky 
I browsed through your code and found there were indeed issues with touch functionality. I've already fixed this problem. You can try replacing the following touch-related code:

```C++
/*Read the touchpad*/
void my_touchpad_read(lv_indev_drv_t *indev_drv, lv_indev_data_t *data)
{
    if (Touch_Int_Flag == true)
    {
        touch.read();
        TP_Point t = touch.getPoint(0);

        if ((touch.getPointNum() == 1) && (t.pressure > 0) && (t.state != 0))
        {
            data->state = LV_INDEV_STATE_PR;

            /*Set the coordinates*/
            data->point.x = t.x;
            data->point.y = t.y;

            // Serial.printf("Touch X: %d Y: %d", t.x, t.y);
            // Serial.printf("Static: %d", t.state);
            // Serial.printf("Pressure: %d", t.pressure);
        }

        Touch_Int_Flag = false;
    }
    else
    {
        data->state = LV_INDEV_STATE_REL;
    }
}
```

8lue5ky, 2024-09-12:
Thank you very much. It works now perfectly fine 👍

closed 2024-09-12 by 8lue5ky