Xinyuan-LilyGO/T-Panel #15: OpenHASP Support

shubham-vunet, 2024-10-31:
### Operating System

OpenHasp

### IDE

PlatformIO

### Software Package Version

-

### Description

I want to use Open HASP and after weeks of debugging I am unable to get screen properly working, After Sometime It doesn't work. I have to reflash LVGL Example then followed by reflash. I made below changes, Please suggest what I am missing.

https://github.com/HASwitchPlate/openHASP/pull/807/files

I am unable to get touch working too.

### Sketch

```cpp
https://github.com/HASwitchPlate/openHASP
```

### Debug Message

```plain
-
```

shubham-vunet, 2024-11-13:
Any update @8lue5ky @Llgok

Llgok, 2024-11-21:
@shubham-vunet 
T-Panel now supports OpenHASP development. For specific example code, please refer to my Fork: [T-Panel openHASP](https://github.com/Llgok/openHASP).

shubham-vunet, 2024-11-21:
Great 👍👍

Llgok, 2024-11-25:
Because of inactivity for an extended period, it was closed.

closed 2024-11-25 by Llgok