Xinyuan-LilyGO/T-Panel #18: Misleading pin definition / overview

jsonslim, 2025-06-15:
The display pins review in the readme section doesn't match the pins on the schematic diagram. Please fix

jsonslim, 2025-06-15:
`<html><body>
<div class="markdown-heading" dir="auto"><h2 tabindex="-1" class="heading-element" dir="auto">PinOverview</h2><a id="user-content-pinoverview" class="anchor" aria-label="Permalink: PinOverview" href="https://github.com/Xinyuan-LilyGO/T-Panel#pinoverview"><svg class="octicon octicon-link" viewBox="0 0 16 16" version="1.1" width="16" height="16" aria-hidden="true"></svg></a></div>
Display Pin | ESP32S3 Pin | XL95X5 Pin
-- | -- | --
VSYNC | ESP32S3_IO40 | NULL
HSYNC | ESP32S3_IO39 | NULL
PCLK | ESP32S3_IO41 | NULL
B0 | ESP32S3_IO1 | NULL
B1 | ESP32S3_IO2 | NULL
B2 | ESP32S3_IO3 | NULL
B3 | ESP32S3_IO4 | NULL
B4 | ESP32S3_IO5 | NULL
G0 | ESP32S3_IO6 | NULL
G1 | ESP32S3_IO7 | NULL
G2 | ESP32S3_IO8 | NULL
G3 | ESP32S3_IO9 | NULL
G4 | ESP32S3_IO10 | NULL
G5 | ESP32S3_IO11 | NULL
R0 | ESP32S3_IO12 | NULL
R1 | ESP32S3_IO13 | NULL
R2 | ESP32S3_IO42 | NULL
R3 | ESP32S3_IO46 | NULL
R4 | ESP32S3_ IO45 | NULL
BL | ESP32S3_IO14 | NULL
CS | NULL | XL95X5_IO17
SCLK | NULL | XL95X5_IO15
MOSI | NULL | XL95X5_IO16
RST | NULL | XL95X5_IO5

</body>
</html>`

Llgok, 2025-06-21:
@jsonslim 
Hello, the pin definition shall be based on the configuration in the software. [pin_config.h](https://github.com/Xinyuan-LilyGO/T-Panel/blob/arduino-esp32-libs_V2.0.14/libraries/Mylibrary/pin_config.h)